kiwih/cubeide-sd-card #1: SCLK speed

carolcabral, 2020-12-03:
Hi!

How did you configure SCLK speeds, if I2SPR register is available only in I2S mode? I'm using STM32F103 but I checked on STM32F303RE user manual and it says: _These bits should be configured when the I2S is disabled. They are used only when the I2S is in master mode. They are not used in SPI mode._

```
#define FCLK_SLOW() { SD_SPI_HANDLE.Instance->I2SPR = 128; }	/* Set SCLK = slow, approx 280 KBits/s*/
#define FCLK_FAST() { SD_SPI_HANDLE.Instance->I2SPR = 8; }	/* Set SCLK = fast, approx 4.5 MBits/s */
```

kiwih, 2020-12-08:
Thanks for bringing this to my attention! You are exactly correct - this appears to have been a bug, and I expect that the library actually was never upping the speed (and instead runs the SD card at the slow speed continuously) (my mistake - I wrote this code originally for an AVR, and this appears to be an error from porting the code over). I've now fixed the bug by changing the lines to:
```c
//(Note that the _256 is used as a mask to clear the prescalar bits as it provides binary 111 in the correct position)
#define FCLK_SLOW() { MODIFY_REG(SD_SPI_HANDLE.Instance->CR1, SPI_BAUDRATEPRESCALER_256, SPI_BAUDRATEPRESCALER_128); }	/* Set SCLK = slow, approx 280 KBits/s*/
#define FCLK_FAST() { MODIFY_REG(SD_SPI_HANDLE.Instance->CR1, SPI_BAUDRATEPRESCALER_256, SPI_BAUDRATEPRESCALER_8); }	/* Set SCLK = fast, approx 4.5 MBits/s */
```

closed 2020-12-08 by kiwih

jessehamner, 2020-12-09:
Yes! That fixed it. Thanks very much. I ran into this issue a few days ago as well.

commit e06d5060 (2020-12-08): Fix code to set SPI prescaler, resolves #1