kiwih/cubeide-sd-card #2: f_mount error (3) when not setting internal pullups on SPI pins

midasgossye, 2021-04-07:
First of all, thanks for the well written how-to guide and blog post! 
Thanks to your explanation and code I managed to read and write to a MicroSD card using a custom STM32F4 dev board 😃 
However, it didn't work first-go and I had to spend a couple of hours to figure out what the issue was...

I have the microSD card directly connected to the SPI2 interface pins on a STM32F405RGT6, like so:
[image]

I altered and uploaded the code and it refused to work, it always gave me a ```f_mount error (3) ``` error...
I hooked up an SD card interface module (like the one you used in your blogpost) to the same SPI pins, and it started working!
After some probing I noticed that the SD card interface module has pullup resistors on all of the data lines. 
After enabling the internal pullups of the SPI interface using the STM32CubeIDE configurator it suddenly started to work!
[image]

This was definitely a goof up on my part, but it would be nice to add a note on your blogpost or code to enable the internal pullups if someone is directly connecting the SD card pins to the STM32. Hopefully, it can save some other people some time so they don't have to go troubleshooting like I did 😅

kiwih, 2021-04-08:
This is great to know. I have added a note to the blog and README.

closed 2021-04-08 by kiwih

zharfanw, 2021-04-16:
I had the same problem too, but it was solved. I was using NUCLEO-64 STM32F446RE, with the same sd card module, and using SD Card Sandisk 16GB. 

I did the same thing with [blog instruction ](https://01001000.xyz/2020-08-09-Tutorial-STM32CubeIDE-SD-card/) and I got error (3). And I solved it with all the SPI2 pins(MOSI, MISO, SCK, and CS) are switched to Pull-Down. Actually, I don't know why? but it's work for me.

Thanks for your code and blog post. it's helpful. I leave a comment here to share my experience with the troubles, but the code it's fine for this far.

zharfanw, 2021-04-16:
> I had the same problem too, but it was solved. I was using NUCLEO-64 STM32F446RE, with the same sd card module, and using SD Card Sandisk 16GB.
>
> I did the same thing with [blog instruction [...]

[image]
[image]

lakshminarayanaks79, 2021-09-07:
Hi this lakshminarayana KS
 i am trying to integrate SD card with SPI over STM32f469I discovery board but i always get the same problem that f_mount error (3)  i followed to blog and also made changed specified here  but nothing has worked for me till now.
could be able to help with latest STM32Cube id

Thanks in advance

With Regards
Lakshminarayana KS

kiwih, 2021-09-09:
Also try changing your SD card @lakshminarayanaks79 , as noted in the blog, not every SD card likes talking SPI.

thibeaux, 2021-09-12:
I fixed this problem by following the steps in this tutorial by Kiwih, https://01001000.xyz/2020-08-09-Tutorial-STM32CubeIDE-SD-card/

You have to make sure you edit the user_diskio.c file adding in the appropriate user code that returns the SPI commands that is used in user_disk_spi.h. This returned the size of my drive being printed into my UART terminal. 

I'm using a SANDISK ULTRA 32GB Micro SD Card, HW-125 SD card SPI adapter for my Nucleo-F446RE. 

Thanks to you guys! Also I'm using pull-ups on all 4 of my SPI1 pins including the CS

MAXH20, 2022-04-18:
Set **FS_TINY** Enabled will make the trick

nelsonifechukwu, 2024-12-10:
Thank you, kiwi. I got mine working. 
I'll take insights about pulling these communication lines high to communicate with a flash memory. It's been a sweat shop fr months now.

pclinh2001, 2025-01-08:
I followed the instructions and it works fine with STM32F1, but when I try with STM32G4 "free_sectors" is always 0 causing "fr_invalid_object" error

Shivabalan012, 2025-05-22:
I getting mount Error 3

caroool7, 2025-06-24:
@thibeaux do you have a picture of the HW-125 SD card SPI adapter? I'm using this sd card module  

[image]

and I'm getting a f_mount error (3). I already tried both the pull-up and pull-down, but I still get the same error. I'm trying to interface it via SPI (SPI3) in STM32U5 B-U585I-IOT02A Discovery kit. 

@kiwih 
I documented the steps I followed and would really appreciate if you could kindly take a look and point out any possible issues or missing parts in my configuration or code.

To make it easier to review, I’ve attached:
     • A full report (PDF) describing my setup and code steps
     • The complete STM32CubeIDE project (ZIP)

[SD_SPI_FATFS_NEW (3).zip](https://github.com/user-attachments/files/20886872/SD_SPI_FATFS_NEW.3.zip)

[SD CARD SPI Interface.pdf](https://github.com/user-attachments/files/20886901/SD.CARD.SPI.Interface.pdf)

cmd05, 2025-07-08:
Worked for me by setting the SPI Pins (SCK, MOSI, MISO) as well as CS to pull-up

MohammadR-Kh, 2025-08-12:
I tried to make this work with Pull_Up and Pull_Down, but I still got the f_mount error (3), but fixed it by powering the SD Card module with 5V, previously I was using 3.3V, make sure to use 5V if you are using a module with logic shifter.