kiwih/cubeide-sd-card #8: Problems running on stm32g0

zzzbatmand, 2021-11-11:
I have changed the include in "user_diskio_spi.c" to use "stm32g0xx_hal.h" and managed to get it to run on an STM32g031F6, it fills quite a bit, but it works.
I then tried to make it run on an STM32g030J6, (should be about the same chip, just smaller) but for some reason, it just resets when it calls "f_mount".
I can trace it to "xchg_spi" in "user_diskio_spi.c". This function somehow fails and makes the chip reset. If I try to step into the function "HAL_SPI_TransmitReceive" in it, it just starts showing me assembly code.
Is there something I can try to test?

zzzbatmand, 2021-11-11:
With a bit bore digging, I can see it is "stm32g0xx_hal_spi" where it is supposed to enable spi.
``` 
/* Check if the SPI is already enabled */
if ((hspi->Instance->CR1 & SPI_CR1_SPE) != SPI_CR1_SPE)
{
  /* Enable SPI peripheral */
  __HAL_SPI_ENABLE(hspi);  <----- This one
}
```  
What is weird, is that it works just fine on the g031 board, even though they use the same code.

kiwih, 2021-11-11:
Connect a debugger and step through the code until it crashes? You should be able to derive the reason.

zzzbatmand, 2021-11-12:
I have run some tests and it seems that the problem lies with the PIN configuration.
The RESET pin is not designed to be used as GPIO, but on the SO8N package (which the J6 is) that pin is used for SPI_SCK, so I am currently trying to disable the RESET function on the chip, as others has had success with this.

kiwih, 2021-11-12:
That would also make sense. Some pins on STM cannot drive signals effectively. You might also consider using different pins if possible.

zzzbatmand, 2021-11-25:
The chip I have chose to use, don't have any other pins for use with SPI, but with help from the STM comunity I managed to disable the reset pin.
The SPI device worked without a problem after doing this, though re-programming of the STM device has become quite difficult, as the reset pin no longer works.

If someone has a simmilar small chip, and a similar problem, this is how I disabled the NRST pin.
I connected the "STM32CubeProgrammer" and connected to the chip, then went into the "REG" tab on the left side.
In there I found a register called "NRST_MODE", setting this to 0x2 will cause it to work as a GPIO, and only internal reset will be possible.
```
00: Reserved
01: Reset Input only: a low level on the NRST pin generates system reset, internal RESET not propagated to the NSRT pin
10: GPIO: standard GPIO pad functionality, only internal RESET possible
11: Bidirectional reset: NRST pin configured in reset input/output mode (legacy mode)
```

closed 2021-11-25 by zzzbatmand