kiwih/cubeide-sd-card #11: f_mount error (1) when wake up from stop mode

CandraSN, 2022-07-07:
I use STM32WL with LoRa and SD Card with your driver. My device use stop mode too. I feel confused when my device wake up from stop mode, then f_mount process fails. The error code that appears is always f_mount error (1).
I don't know what parameters need to be initialized again for SD Card system to work when wake up from stop mode. Maybe there is some configuration i'm missing, please help me.

Thank you

utkarshsethi, 2023-01-25:
not sure if relevant, but SD cards if gone to sleep state need a power cycle to be working again.

try if that's the issue for you.

TronBoy22, 2025-06-05:
I have a STM32L433 and I encountered the same issue in STOP 1 mode. I also had the issue if I was mounting/unmounting multiple times (without using STOP 1 mode). 

The issue was narrowed down to the DSTATUS disk_initialize(BYTE pdrv) function in diskio.c which calls the DSTATUS USER_SPI_initialize(BYTE drv) function in user_diskio_spi.c.
It works the first time because disk.is_initialized[pdrv] == 0, but fails on the next call because it equals 1 and therefore does not call USER_SPI_initialize(). I just set it to 0 each time disk_initialize() is called and it worked.