kiwih/cubeide-sd-card #13: l_seek does not work because f_size always returns 0

JustEnoughDucks, 2023-01-13:
Hello,

I have been trying to get functionality of using l_seek with F_size to get the end of a line since this version (and the HAL version for F0 chips in general) only supports up to R0.11 of FatFS which doesn't include the FA_APPEND flag. 

No matter how much I write to the SD card, even when the files are verified off-system, f_size() function and indeed the fsize member of the file handler is always 0. This makes writing to the end of a file after file close extremely difficult. All other reading and writing functions work correctly. Indeed if I never close the file, I can write many lines without a problem, but for the use-case of having a removable and re-insertable SD card it is a huge problem.

Is this a reproducible error on your side?

kiwih, 2023-01-13:
Not an issue I have ever faced (I don't tend to use the F0 chips). Can you import your own version of FatFS instead of using the HAL-provided version?

JustEnoughDucks, 2023-01-16:
I did try that to move to R0.12c instead of R0.11, but CubeIDE kept on giving me errors that the ff_genb_drv file didn't have any standard types available in the stdlib.h and integer.h that it has an include for. It seems to be optimizing out the essential files, but if I turn optimization off, then the program blows up and I run out of flash because of bloated HAL I2C and SPI libraries.