kiwih/cubeide-sd-card #18: Reading works, writing doesn't (code ported to STM32L4)

Bruno-B2S, 2023-07-19:
Hi, I've followed the guide in the blog and ported your code to STM32L4. I had to enable the internal pull-ups of MISO, MOSI and CLK. Read works, but write doesn't. I've tried with several sd cards from different suppliers and sizes. The file write.txt never shows on the sd card, even if I create an empty one in advance I don't see the string written there. With 80MHz system clock I'm using SPI_BAUDRATEPRESCALER_256 for FCLK_SLOW and SPI_BAUDRATEPRESCALER_16 for FCLK_FAST.

This is the debug output of my program:

```
[I:control.c,200] SD Card test
[I:control.c,230] SD card stats:
    119762 KiB total drive space.
    119754 KiB available.
[I:control.c,238] I was able to open 'test.txt' for reading!
[I:control.c,246] Read string from 'test.txt' contents: Hey!
[I:control.c,271] Wrote 19 bytes to 'write.txt'!
```

I even added a FRES check for closing files and unmounting the filesystem. The functions always return FR_OK.

Any idea about what the problem could be? Anything else that should be changed when porting to STM32L4 (apart from #include "stm32l4xx_hal.h")?

Regards

kiwih, 2023-07-20:
Honestly not sure. In my experience it either all worked or it didn't, not a kind of partial success like this. I note that 119762KiB is not particularly large. Are you using a small SD card or does it have some kind of strange partitions?

kiwih, 2023-07-20:
What happens if you swap the order of the read/write to write first instead of reading?

Bruno-B2S, 2023-07-20:
Sorry, it was a dumb mistake on my side: while I was following your blog post, I overlooked one of the lines marked as "add this line" for the user_diskio.c (specifically USER_write calling USER_SPI_write). I downloaded the file from Git Hub and compared it one-to-one with stm32cubeide. Sorry for wasting your time.

BTW: I was using a small SD card I bought (128MB) so that the sector size matches the maximum available sector size in Cubemx for FatFS. But now I've successfully tested this with 128MB and with 32GB. What might not work is the fs_mkfs function.

closed 2023-07-20 by Bruno-B2S